lisaac/luci-app-diskman #10: 如果把你diskman的插件加入到云编译？

ywgf88, 2020-02-03:
我直接按照你的语句把diskman加入到云编译的脚本，但是每次编译要么报错；要不就没有diskman

git clone https://github.com/lisaac/luci-app-diskman package/luci-app-diskman
mkdir -p package/parted && cp -i package/luci-app-diskman/Parted.Makefile package/parted/Makefile

lisaac, 2020-02-03:
CONFIG_PACKAGE_luci-app-diskman=y

ywgf88, 2020-02-03:
mkdir -p package/parted && cp -i package/luci-app-diskman/Parted.Makefile package/parted/Makefile
上面的语句需要加入吗？ 还是直接两步：
1. git clone https://github.com/lisaac/luci-app-diskman package/luci-app-diskman

2. CONFIG_PACKAGE_luci-app-diskman=y

lisaac, 2020-02-03:
需要的

另外报错内容是什么

ywgf88, 2020-02-03:
1. 我在diy.sh里加入如下命令：
# uci-app-dockerman && diskman
git clone https://github.com/lisaac/luci-lib-docker.git package/luci-lib-docker
git clone https://github.com/lisaac/luci-app-dockerman.git package/luci-app-dockerman
git clone https://github.com/lisaac/luci-app-diskman /package/luci-app-diskman

2. 云编译报错内容：
  Configuration Customization - Build_RaspberryPi10s
##[error]Process completed with exit code 128.
Run [ -e $CONFIG_FILE ] && mv $CONFIG_FILE openwrt/.config
Cloning into 'package/luci-theme-Light'...
Cloning into 'package/diy/luci-theme-Night'...
Cloning into 'package/luci-theme-opentomcat'...
Cloning into 'package/luci-theme-leo'...
Cloning into 'package/luci-app-vssr-coexist'...
Cloning into 'package/openwrt-package'...
Cloning into 'package/luci-lib-docker'...
Cloning into 'package/luci-app-dockerman'...
fatal: could not create leading directories of '/package/luci-app-diskman': Permission denied
##[error]Process completed with exit code 128.

lisaac, 2020-02-03:
`package`前多了一个`/`

另外你没加
```
mkdir -p package/parted && cp -i package/luci-app-diskman/Parted.Makefile package/parted/Makefile
```
也没有修改.config

ywgf88, 2020-02-03:
1. 已经去掉/
2.已经添加mkdir -p package/parted && cp -i package/luci-app-diskman/Parted.Makefile package/parted/Makefile
3. 已经在.config里添加CONFIG_PACKAGE_luci-app-diskman=y
4. 重试一遍。

等待结果，待会有结果通知你，谢谢！！！

closed 2020-02-04 by lisaac