lisaac/luci-app-diskman #22: Bad translation of 3 words

IceG2020, 2020-03-09:
Hi :)
Recently I noticed that I do not have a translation of the phrase "Free Space" in the "Partition info" window, I added a translation to the file "diskman.po" but nothing changes why, I have no translation for "Free Space" and another translation for "Used" and "Usage". What could be wrong? :)
Regards
IceG

lisaac, 2020-03-09:
Is it work now?

closed 2020-03-10 by lisaac

IceG2020, 2020-03-10:
[image]

Hi :)
Unfortunately, it still displays differently than it is in the translation file .. strange ..

lisaac, 2020-03-10:
I believe that you have another luci app has translated these 3 word, then luci use that translate word to display on diskman, please check.
Thanks

IceG2020, 2020-03-10:
From all applications installed on my router only luci-app-opkg has the phrase "Free space:" and it is translated correctly.
Regards

lisaac, 2020-03-10:
It's weird, since the another lanuage is worked.
I still believe there is another app has translated it.
Simultaneously, we using 'Free Space', not 'Free space:'

Please check it again.

IceG2020, 2020-03-12:
Everything is ok, I cleaned the compilation and the new build image has everything beautifully translated.
Regards
IceG