lisaac/luci-app-diskman #34: 大佬，编译有问题了

tcsr200722, 2020-11-28:
mkdir -p package/luci-app-diskman && \
wget https://raw.githubusercontent.com/lisaac/luci-app-diskman/master/Makefile -O package/luci-app-diskman/Makefile
mkdir -p package/parted && \
wget https://raw.githubusercontent.com/lisaac/luci-app-diskman/master/Parted.Makefile -O package/parted/Makefile

#compile package only
make package/luci-app-diskman/compile V=99

#compile
make menuconfig
#choose LuCI ---> 3. Applications  ---> <*> luci-app-diskman..... Disk Manager interface for LuCI ----> save
make V=99

按照这个编译，编译不出来了

lisaac, 2020-11-29:
路径调整了一下，需要修改下readme

tcsr200722, 2020-11-29:
好的 谢谢

closed 2020-11-29 by tcsr200722

tcsr200722, 2020-11-30:
make[3]: Entering directory '/workdir/openwrt/package/luci-app-diskman'
find: '/workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/po/*/*.po': No such file or directory
rm -rf /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman.installed /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman
mkdir -p /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman
# install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/
# cp -pR /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/root/* /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/
# install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/www
# cp -pR /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/htdoc/* /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/www
install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/usr/lib/lua/luci
cp -pR /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/luasrc/* /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/usr/lib/lua/luci/
cp: cannot stat '/workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/luasrc/*': No such file or directory
make[3]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman.installed] Error 1
Makefile:85: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman.installed' failed
make[3]: Leaving directory '/workdir/openwrt/package/luci-app-diskman'
make[2]: *** [package/luci-app-diskman/compile] Error 2
time: package/luci-app-diskman/compile#0.13#0.05#0.19
package/Makefile:111: recipe for target 'package/luci-app-diskman/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
Error: Process completed with exit code 2.

hoatienii, 2020-12-13:
same error

make[2]: Entering directory '/home/hass/openwrt/package/utils/mdadm'
echo "mdadm" >> /home/hass/openwrt/staging_dir/target-x86_64_musl/pkginfo/mdadm.default.install
make[2]: Leaving directory '/home/hass/openwrt/package/utils/mdadm'
time: package/utils/mdadm/compile#0.32#0.10#0.42
make[2]: Entering directory '/home/hass/openwrt/package/luci-app-diskman/applications/luci-app-diskman'
find: '/home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/po/*/*.po': No such file or directory
rm -rf /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman.installed /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman
mkdir -p /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman
# install -d -m0755 /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/
# cp -pR /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/root/* /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/
# install -d -m0755 /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/www
# cp -pR /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/htdoc/* /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/www
install -d -m0755 /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/usr/lib/lua/luci
cp -pR /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/luasrc/* /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/usr/lib/lua/luci/
cp: cannot stat '/home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/luasrc/*': No such file or directory
make[2]: *** [Makefile:92: /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman.installed] Error 1
make[2]: Leaving directory '/home/hass/openwrt/package/luci-app-diskman/applications/luci-app-diskman'
time: package/luci-app-diskman/applications/luci-app-diskman/compile#0.22#0.11#0.32
make[1]: *** [package/Makefile:113: package/luci-app-diskman/applications/luci-app-diskman/compile] Error 2
make[1]: Leaving directory '/home/hass/openwrt'
make: *** [/home/hass/openwrt/include/toplevel.mk:236: package/luci-app-diskman/compile] Error 2