lisaac/luci-app-diskman #35: 大佬，编译还是存在问题

tcsr200722, 2020-12-05:
install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/usr/lib/lua/luci
cp -pR /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/luasrc/* /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/usr/lib/lua/luci/
cp: cannot stat '/workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/luasrc/*': No such file or directory
make[3]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman.installed] Error 1
Makefile:85: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman.installed' failed
make[3]: Leaving directory '/workdir/openwrt/package/luci-app-diskman'
make[2]: *** [package/luci-app-diskman/compile] Error 2
time: package/luci-app-diskman/compile#0.13#0.05#0.19
package/Makefile:111: recipe for target 'package/luci-app-diskman/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
Error: Process completed with exit code 2.

hoatienii, 2020-12-13:
same error!

make[2]: Entering directory '/home/hass/openwrt/package/utils/mdadm'
echo "mdadm" >> /home/hass/openwrt/staging_dir/target-x86_64_musl/pkginfo/mdadm.default.install
make[2]: Leaving directory '/home/hass/openwrt/package/utils/mdadm'
time: package/utils/mdadm/compile#0.32#0.10#0.42
make[2]: Entering directory '/home/hass/openwrt/package/luci-app-diskman/applications/luci-app-diskman'
find: '/home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/po/*/*.po': No such file or directory
rm -rf /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman.installed /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman
mkdir -p /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman
# install -d -m0755 /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/
# cp -pR /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/root/* /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/
# install -d -m0755 /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/www
# cp -pR /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/htdoc/* /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/www
install -d -m0755 /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/usr/lib/lua/luci
cp -pR /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/luasrc/* /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman/usr/lib/lua/luci/
cp: cannot stat '/home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/luasrc/*': No such file or directory
make[2]: *** [Makefile:92: /home/hass/openwrt/build_dir/target-x86_64_musl/luci-app-diskman/applications/luci-app-diskman/.pkgdir/luci-app-diskman.installed] Error 1
make[2]: Leaving directory '/home/hass/openwrt/package/luci-app-diskman/applications/luci-app-diskman'
time: package/luci-app-diskman/applications/luci-app-diskman/compile#0.22#0.11#0.32
make[1]: *** [package/Makefile:113: package/luci-app-diskman/applications/luci-app-diskman/compile] Error 2
make[1]: Leaving directory '/home/hass/openwrt'
make: *** [/home/hass/openwrt/include/toplevel.mk:236: package/luci-app-diskman/compile] Error 2

obsy, 2020-12-16:
Wrong path after https://github.com/lisaac/luci-app-diskman/commit/8a2c534e549d8a4786b30ccb02df813e0c324b0b

Solution:
```
--- a/applications/luci-app-diskman/Makefile
+++ b/applications/luci-app-diskman/Makefile
@@ -74,9 +74,9 @@ define Package/$(PKG_NAME)/install
        # $(INSTALL_DIR) $(1)/www
        # cp -pR $(PKG_BUILD_DIR)/htdoc/* $(1)/www
        $(INSTALL_DIR) $(1)/usr/lib/lua/luci
-       cp -pR $(PKG_BUILD_DIR)/luasrc/* $(1)/usr/lib/lua/luci/
+       cp -pR $(PKG_BUILD_DIR)/../../luasrc/* $(1)/usr/lib/lua/luci/
        $(INSTALL_DIR) $(1)/usr/lib/lua/luci/i18n
-       $$(foreach po,$$(shell find $(PKG_BUILD_DIR)/po/*/*.po), \
+       $$(foreach po,$$(shell find $(PKG_BUILD_DIR)/../../po/*/*.po), \
                po2lmo $$(po) \
                $(1)/usr/lib/lua/luci/i18n/diskman.$$(shell echo $$(po) | awk -F'/' '{print $$$$(NF-1)}').lmo;)
        #po2lmo $(PKG_BUILD_DIR)/po/zh-cn/diskman.po $(1)/usr/lib/lua/luci/i18n/diskman.zh-cn.lmo
```

hoatienii, 2020-12-17:
> $$(foreach po,$$(shell find $(PKG_BUILD_DIR)/../../po/*/*.po), \

Thanks so much! I will try.

lisaac, 2020-12-18:
have fixed

closed 2020-12-18 by lisaac