lisaac/luci-app-diskman #36: Wrong size of disk "bar"

obsy, 2021-02-26:
[image]

For 2TB drive with one partition - "bar" is 700% wide. Info about drive:
```
{path: "/dev/sda",…}
extended_partition_index: 2
flags: ""
health: "PASSED"
logic_sec: "512"
model: "ATA ST2000DM001-1CH1"
p_table: "MBR"
partitions: [{number: -1, sec_start: "2", fs: "Free Space", sec_end: "2047", size_formated: "1023.00 KB",…},…]
0: {number: -1, sec_start: "2", fs: "Free Space", sec_end: "2047", size_formated: "1023.00 KB",…}
flags: ""
free: 0
free_formated: "-"
fs: "Free Space"
mount_point: "-"
name: "-"
number: -1
sec_end: "2047"
sec_start: "2"
size: 1047552
size_formated: "1023.00 KB"
tag_name: ""
usage: "-"
used: 0
used_formated: "-"
1: {number: 1, used_formated: "835.95 MB", free: 1967032442880, size_formated: "1.82 TB", tag_name: "",…}
flags: ""
free: 1967032442880
free_formated: "1.79 TB"
fs: "-"
logicals: []
mount_point: "/tmp/dysk"
name: "sda1"
number: 1
sec_end: "3907029167"
sec_start: "2048"
size: 16003183083520
size_formated: "1.82 TB"
tag_name: ""
type: "extended"
usage: "0%"
used: 876560384
used_formated: "835.95 MB"
path: "/dev/sda"
phy_sec: "4096"
rota_rate: "7200 rpm"
sata_ver: "SATA 3.1, 6.0 Gb/s (current: 3.0 Gb/s)"
sec_size: "512/4096"
size: "2000398934016"
size_formated: "1.82 TB"
sn: "Z1E469CT"
status: "ACTIVE"
temp: "46°C"
type: "scsi"
```

Wrong part1 size? (16003183083520)

lisaac, 2021-03-05:
@obsy 
yes, it seems the part1 size is wrong, you may check your disk first.

obsy, 2021-03-05:
Disk is fine. Problem is with this code: https://github.com/lisaac/luci-app-diskman/blob/master/applications/luci-app-diskman/luasrc/model/diskman.lua#L229

My drive has MBR (!). First partition is primary (number 1), counting as 512byte sector == 1.8TB. Not extended, with phisical size 4096. See:
```
root@OpenWrt:~# parted /dev/sda  print free
Model: ATA ST2000DM001-1CH1 (scsi)
Disk /dev/sda: 2000GB
Sector size (logical/physical): 512B/4096B
Partition Table: msdos
Disk Flags: 

Number  Start   End     Size    Type     File system  Flags
        1024B   1049kB  1048kB           Free Space
 1      1049kB  2000GB  2000GB  primary  ext4

root@OpenWrt:~# parted /dev/sda unit s print free 
Model: ATA ST2000DM001-1CH1 (scsi)
Disk /dev/sda: 3907029168s
Sector size (logical/physical): 512B/4096B
Partition Table: msdos
Disk Flags: 

Number  Start  End          Size         Type     File system  Flags
        2s     2047s        2046s                 Free Space
 1      2048s  3907029167s  3907027120s  primary  ext4

root@OpenWrt:~# parted -m /dev/sda unit s print free 
BYT;
/dev/sda:3907029168s:scsi:512:4096:msdos:ATA ST2000DM001-1CH1:;
1:2s:2047s:2046s:free;
1:2048s:3907029167s:3907027120s:ext4::;
root@OpenWrt:~# 
```

But this fragment of code recognized this partition as extended with size ~ 3907xxxx * 4096. Should be 512.