lisaac/luci-app-diskman #42: 无法正常获取 S.M.A.R.T. 显示状态为 SLEEP

love4taylor, 2021-08-06:
https://github.com/lisaac/luci-app-diskman/blob/b6b499a08f1bd0e420211adbc8331d8e59243b4a/applications/luci-app-diskman/luasrc/model/diskman.lua#L39

使用了 `-n standby` 时 VL716 硬盘盒+SSD 状态显示为 SLEEP，无法正常获取 S.M.A.R.T. 但用 hdparm -C 显示 `drive state is:  standby`

```
root@ImmortalWrt:~# smartctl -H -A -i -n standby -f brief /dev/sda
smartctl 7.2 2020-12-30 r5155 [aarch64-linux-5.4.137] (localbuild)
Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org

Device is in SLEEP mode, exit(2)
```

lisaac, 2021-08-12:
这个应该是硬盘盒的原因，大多数usb硬盘smartctl都无法读取smart信息的

love4taylor, 2021-08-12:
读是没问题的，只要去掉 `-n standby` 就能正常显示了（需要安装了 smartmontools-drivedb）现在主要的问题就是会被 smartctl 判断为 SLEEP。

lisaac, 2021-08-12:
> 读是没问题的，只要去掉 `-n standby` 就能正常显示了（需要安装了 smartmontools-drivedb）现在主要的问题就是会被 smartctl 判断为 SLEEP。

这就尴尬了，没研究过smartctl，如果去掉`-n standby`，会让处于休眠的硬盘唤醒，带来不必要的唤醒

lisaac, 2021-08-16:
@love4taylor 
貌似解决方案找到了？

love4taylor, 2021-08-16:
并没有 另一个 issue 的是另一个硬盘盒

closed 2026-10-08 by love4taylor