lisaac/luci-app-diskman #44: 外置 NVME 盘在luci界面里无法获得温度

love4taylor, 2021-08-15:
smartctl 命令是可以拿到温度信息的 `Temperature:                        33 Celsius`，luci 里没显示。

```
root@Sakura:~# smartctl -H -A -i -n standby -f brief /dev/sda
smartctl 7.2 2020-12-30 r5155 [aarch64-linux-5.4.138] (localbuild)
Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       INTEL SSDPEKKF010T8L
Serial Number:                      ****************
Firmware Version:                   L14P
PCI Vendor/Subsystem ID:            0x8086
IEEE OUI Identifier:                0x5cd2e4
Controller ID:                      1
NVMe Version:                       1.3
Number of Namespaces:               1
Namespace 1 Size/Capacity:          1,024,209,543,168 [1.02 TB]
Namespace 1 Formatted LBA Size:     512
Namespace 1 IEEE EUI-64:            5cd2e4 2301a0012f
Local Time is:                      Mon Aug 16 01:40:06 2021 CST

=== START OF SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

SMART/Health Information (NVMe Log 0x02)
Critical Warning:                   0x00
Temperature:                        33 Celsius
Available Spare:                    100%
Available Spare Threshold:          12%
Percentage Used:                    0%
Data Units Read:                    21,056,019 [10.7 TB]
Data Units Written:                 5,265,230 [2.69 TB]
Host Read Commands:                 103,817,820
Host Write Commands:                32,322,631
Controller Busy Time:               734
Power Cycles:                       3,341
Power On Hours:                     377
Unsafe Shutdowns:                   243
Media and Data Integrity Errors:    0
Error Information Log Entries:      0
Warning  Comp. Temperature Time:    0
Critical Comp. Temperature Time:    0
```

[image]

lisaac, 2021-08-16:
@love4taylor 
打个补丁试试：
```
sed 's/if dev:match("nvme")then/if cmd:match("NVMe Version:")then/' /tmp/.luci/usr/lib/lua/luci/
controller/diskman.lua
rm -fr /tmp/luci-*
```

love4taylor, 2021-08-16:
不行 还是空的

```
root@Sakura:/sys/bus/usb/devices# grep NVMe /usr/lib/lua/luci/controller/diskman.lua
    if cmd:match("NVMe Version:")then
```

lisaac, 2021-08-16:
> 不行 还是空的
> 
> ```
> root@Sakura:/sys/bus/usb/devices# grep NVMe /usr/lib/lua/luci/controller/diskman.lua
>     if cmd:match("NVMe Version:")then
> ```

luci 缓存清了吗 
`rm -fr /tmp/luci-*`

love4taylor, 2021-08-16:
清了的

lisaac, 2021-08-16:
> 清了的

能否看下浏览器 XHR 返回的数据

love4taylor, 2021-08-16:
```
{
    "path": "/dev/sda",
    "partitions": [
        {
            "number": -1,
            "sec_start": "2",
            "fs": "Free Space",
            "sec_end": "2047",
            "size_formated": "1023.00 KB",
            "tag_name": "",
            "used_formated": "-",
            "flags": "",
            "mount_point": "-",
            "name": "-",
            "free": 0,
            "used": 0,
            "usage": "-",
            "free_formated": "-",
            "size": 1047552
        },
        {
            "number": 1,
            "sec_start": "2048",
            "type": "primary",
            "fs": "ext4",
            "sec_end": "2000408575",
            "size_formated": "953.87 GB",
            "tag_name": "",
            "used_formated": "116.72 GB",
            "flags": "boot",
            "mount_point": "/mnt/sda1",
            "name": "sda1",
            "free": 830419722240,
            "used": 125332508672,
            "usage": "13%",
            "free_formated": "773.39 GB",
            "size": 1024208142336
        },
        {
            "number": -1,
            "sec_start": "2000408576",
            "fs": "Free Space",
            "sec_end": "2000409263",
            "size_formated": "344.00 KB",
            "tag_name": "",
            "used_formated": "-",
            "flags": "",
            "mount_point": "-",
            "name": "-",
            "free": 0,
            "used": 0,
            "usage": "-",
            "free_formated": "-",
            "size": 352256
        }
    ],
    "logic_sec": "512",
    "sec_size": "512/512",
    "sn": "***********",
    "flags": "",
    "size_formated": "953.87 GB",
    "health": "PASSED",
    "model": "INTEL SS DPEKKF010T8L",
    "p_table": "MBR",
    "type": "scsi",
    "phy_sec": "512",
    "size": "1024209543168"
}
```

lisaac, 2021-08-16:
请测试下这个提交 https://github.com/lisaac/luci-app-diskman/commit/6c5451c95cb2b3bfffb702b9fee814d852f462b1

love4taylor, 2021-08-16:
可以了

[image]

closed 2021-08-16 by love4taylor