lisaac/luci-app-diskman #45: 编辑界面里的健康点进去 500 了

love4taylor, 2021-08-16:
[image]

```
/usr/lib/lua/luci/controller/diskman.lua:113: attempt to call method 'match' (a nil value)
stack traceback:
	/usr/lib/lua/luci/controller/diskman.lua:113: in function </usr/lib/lua/luci/controller/diskman.lua:108>
```

lisaac, 2021-08-16:
改下smartctl 命令看看
```
local cmd = io.popen(dm.command.smartctl ..  " -H -A -i /dev/%s" % dev)
```

love4taylor, 2021-08-16:
依旧 500

```
root@Sakura:~# grep "dm.command.smartctl" /usr/lib/lua/luci/controller/diskman.lua
  local cmd = io.popen(dm.command.smartctl ..  " -H -A -i /dev/%s" % dev)
```

lisaac, 2021-08-16:
命令行运行下smartctl呢

love4taylor, 2021-08-17:
```
root@Sakura:~# smartctl -H -A -i /dev/sda
smartctl 7.2 2020-12-30 r5155 [aarch64-linux-5.10.56] (localbuild)
Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       INTEL SSDPEKKF010T8L
Serial Number:                      *************
Firmware Version:                   L14P
PCI Vendor/Subsystem ID:            0x8086
IEEE OUI Identifier:                0x5cd2e4
Controller ID:                      1
NVMe Version:                       1.3
Number of Namespaces:               1
Namespace 1 Size/Capacity:          1,024,209,543,168 [1.02 TB]
Namespace 1 Formatted LBA Size:     512
Namespace 1 IEEE EUI-64:            5cd2e4 2301a0012f
Local Time is:                      Tue Aug 17 01:36:16 2021 UTC

=== START OF SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

SMART/Health Information (NVMe Log 0x02)
Critical Warning:                   0x00
Temperature:                        30 Celsius
Available Spare:                    100%
Available Spare Threshold:          12%
Percentage Used:                    0%
Data Units Read:                    21,484,675 [11.0 TB]
Data Units Written:                 5,619,473 [2.87 TB]
Host Read Commands:                 117,274,437
Host Write Commands:                43,545,226
Controller Busy Time:               770
Power Cycles:                       3,351
Power On Hours:                     401
Unsafe Shutdowns:                   253
Media and Data Integrity Errors:    0
Error Information Log Entries:      0
Warning  Comp. Temperature Time:    0
Critical Comp. Temperature Time:    0
```

lisaac, 2021-08-17:
这就没道理了，暂时找不出原因

jjm2473, 2022-06-16:
@lisaac 
io.popen的返回值不是字符串，不支持match方法

https://github.com/lisaac/luci-app-diskman/blob/6ba3005ebdf1faabc7e0c4889c95caa3c153cafb/applications/luci-app-diskman/luasrc/controller/diskman.lua#L113

21.02.2固件中也是500错误

jjm2473, 2022-06-17:
在 https://github.com/lisaac/luci-app-diskman/pull/53 修复了

closed 2022-06-17 by lisaac