lisaac/luci-app-diskman #58: OpenWrt 21.02.3无法打开页面

493505110, 2022-08-12:
```
/usr/lib/lua/luci/dispatcher.lua:1495: module 'luci.cbi' not found:
	no field package.preload['luci.cbi']
	no file './luci/cbi.lua'
	no file '/usr/share/lua/luci/cbi.lua'
	no file '/usr/share/lua/luci/cbi/init.lua'
	no file '/usr/lib/lua/luci/cbi.lua'
	no file '/usr/lib/lua/luci/cbi/init.lua'
	no file './luci/cbi.so'
	no file '/usr/lib/lua/luci/cbi.so'
	no file '/usr/lib/lua/loadall.so'
	no file './luci.so'
	no file '/usr/lib/lua/luci.so'
	no file '/usr/lib/lua/loadall.so'
stack traceback:
	[C]: in function 'require'
	/usr/lib/lua/luci/dispatcher.lua:1495: in function '_form'
	/usr/lib/lua/luci/dispatcher.lua:1027: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:1000: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478>
```
版本: OpenWrt 21.02.3 r16554-1d4dea6d4f

lisaac, 2022-08-13:
添加 luci-compat 依赖

493505110, 2022-08-15:
添加依赖后可以正常使用了，谢谢

closed 2022-08-15 by 493505110

Hurricos, 2022-11-24:
Why not have an explicit dependency on `luci-compat`, then?