lisaac/luci-app-diskman #60: makefile需要改进，有导致依赖不会被自动关联上的问题，导致编译出来的固件没有依赖

wlshdjj, 2022-08-20:
下面是需要改动的部分：
#--------------------------------------------------------------------------
include $(TOPDIR)/rules.mk

PKG_NAME:=luci-app-diskman
PKG_VERSION:=v0.2.11
PKG_RELEASE:=beta

PKG_LICENSE:=GPL-3.0-only
PKG_MAINTAINER:=lisaac <https://github.com/lisaac/luci-app-diskman>

PKG_CONFIG_DEPENDS:= \
	CONFIG_PACKAGE_$(PKG_NAME)_INCLUDE_btrfs_progs \
	CONFIG_PACKAGE_$(PKG_NAME)_INCLUDE_lsblk \
	CONFIG_PACKAGE_$(PKG_NAME)_INCLUDE_mdadm \
	CONFIG_PACKAGE_$(PKG_NAME)_INCLUDE_kmod_md_raid456 \
	CONFIG_PACKAGE_$(PKG_NAME)_INCLUDE_kmod_md_linears

LUCI_TITLE:=Disk Manager interface for LuCI
LUCI_DEPENDS:=+e2fsprogs +parted +smartmontools +blkid \
	+PACKAGE_$(PKG_NAME)_INCLUDE_btrfs_progs:btrfs-progs \
	+PACKAGE_$(PKG_NAME)_INCLUDE_lsblk:lsblk \
	+PACKAGE_$(PKG_NAME)_INCLUDE_mdadm:mdadm \
	+PACKAGE_$(PKG_NAME)_INCLUDE_kmod_md_raid456:mdadm \
	+PACKAGE_$(PKG_NAME)_INCLUDE_kmod_md_raid456:kmod-md-raid456 \
	+PACKAGE_$(PKG_NAME)_INCLUDE_kmod_md_linears:mdadm \
	+PACKAGE_$(PKG_NAME)_INCLUDE_kmod_md_linears:kmod-md-linear
LUCI_PKGARCH:=all
#--------------------------------------------------------------------------

wlshdjj, 2022-08-20:
另外原来的makefile每次编译都要强行git一次，实在无此必要，有时候git失败导致编译失败，浪费很多时间

jjm2473, 2023-02-04:
可以使用我的这个分支，改成了标准的luci makefile，不会再从git克隆 https://github.com/jjm2473/luci-app-diskman/tree/dev

lisaac, 2023-02-04:
期待 PR

> 可以使用我的这个分支，改成了标准的luci makefile，不会再从git克隆 https://github.com/jjm2473/luci-app-diskman/tree/dev

jjm2473, 2023-02-05:
@lisaac 你的原始makefile这样写不是为了方便某些人单独拷贝makefile就可以编译吗？如果改成标准的makefile，那就需要加feed才能编译。不知道是不是你想要的

lisaac, 2023-02-13:
> @lisaac 你的原始makefile这样写不是为了方便某些人单独拷贝makefile就可以编译吗？如果改成标准的makefile，那就需要加feed才能编译。不知道是不是你想要的

改成标准的也行的

PR #65 (2023-02-26): fix some issues
fixed #60
fixed #59
commit a7d27239 (2023-02-26): Merge pull request #65 from muink/master

fixed #60
fixed #59
closed 2023-02-26 by lisaac