lisaac/luci-app-diskman #67: 随固件编译，luci里没有界面

Speechless22, 2023-04-01:
[image]
依赖都正常

Speechless22, 2023-04-01:
前两天直接装你的ipk还能用，编到固件里就没界面了。从官方master分支编译

muink, 2023-04-21:
应该是因为没有写下面两个文件, 现在openwrt官方源21.02开始, 用标准编译方式, 不写 `luci/menu.d` 是没有界面的
暂时的解决方法是卸载新版用老版本升级上来
`/usr/share/luci/menu.d/luci-app-diskman.json`
`/usr/share/rpcd/acl.d/luci-app-tinyproxy.json`

muink, 2023-04-22:
重现了下错误, 是浏览器缓存的问题, 开启 `F12` 开发者工具, 在分页 `Network` 里把 `Disable cache` 勾选上刷新页面就好了
[image]

muink, 2023-04-22:
@wazzx 我这边已经无法复现问题了, 应该是正常了
你看看会不会是 `luci-compat` `luci-lib-ipkg` 这两个包没装
或者用我这边编译的试试 [luci-app-diskman_v0.2.12_all.ipk](https://fantastic-packages.github.io/packages/releases/21.02/packages/x86_64/packages/luci-app-diskman_v0.2.12_all.ipk)

muink, 2023-04-22:
2KB确实不正常, 可能是编译的步骤出了什么问题, 我这里编译出来是 21KB
你可以下载上个对话里的ipk看看, 是能正常用的
[image]

muink, 2023-04-22:
luci-app是全平台通用, 你可以下载来直接用

muink, 2023-04-22:
卸载掉新版本装[老版本](https://github.com/lisaac/luci-app-diskman/releases/tag/v0.2.11)试试呢
界面会有吗, 有的话再时间安装新版看看

Nelocompne, 2023-08-26:
> [image]
> 
> @wazzx 我这边已经无法复现问题了, 应该是正常了 你看看会不会是 `luci-compat` `luci-lib-ipkg` 这两个包没装 或者用我这边编译的试试 [luci-app-diskman_v0.2.12_all.ipk](https://fantastic-packages.github.io/packages/releases/21.02/packages/x86_64/packages/luci-app-diskman_v0.2.12_all.ipk) v0.2.13版本还在编译, 就修复了个没有编译语言文件的问题 估计再一两个小时自动推送到 https://fantastic-packages.github.io/packages

我的固件也是无法看到按钮，但是装上  `luci-compat` `luci-lib-ipkg`  两个包后就能看到了，这对我很有帮助！

closed 2023-10-08 by Speechless22