lisaac/luci-app-diskman #74: wget: not found

jingaoyong, 2023-11-02:
BusyBox v1.33.2 (2023-07-13 04:40:21 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 21.02-SNAPSHOT
 -----------------------------------------------------
root@GL-MT3000:~# mkdir -p package/luci-app-diskman && \ wget https://raw.git
hubusercontent.com/lisaac/luci-app-diskman/master/applications/luci-app-diskm
an/Makefile -O package/luci-app-diskman/Makefile
-ash:  wget: not found

________________________
纯小白不会编辑，只能用大神的，而且还看不懂。只会复制粘贴但也想借大佬的东风搭上高速列车，结果。。。。还没起飞就夭折了 😂

muink, 2023-11-02:
这个命令是编译用的
如果是要在路由器上安装的话
https://fantastic-packages.github.io/packages/releases/21.02/packages/aarch64_cortex-a53/luci/
这里下载

jingaoyong, 2023-11-03:
[image]
[image]
谢谢大佬回复！安装好了，但是看见路由器里面的diskman和大佬项目介绍中的截图不一样，不知道如何使用它来管理磁盘。 想给路由器的overlay 增加空间，已经满了，尴尬