lisaac/luci-app-diskman #75: 安装后没有显示菜单入口，软件不能使用

jim-sang, 2023-12-16:
日志中有一条
Lua controller /usr/lib/lua/luci/controller/diskman.lua present but no Lua runtime installed.

muink, 2023-12-16:
Try `opkg update && opkg install --forcereinstall luci-lua-runtime luci-compat luci-lib-ipkg`

sunowsir, 2024-09-04:
同样现象，系统日志显示： daemon.err uhttpd[2731]: /usr/bin/which: no parted in ((null))
解决办法：
1. 编辑 /usr/lib/lua/luci/controller/diskman.lua， 修改约17行，local command = luci.sys.exec("/usr/bin/which" .. cmd) 为 local command = luci.sys.exec("export PATH=$PATH:/sbin:/usr/sbin && /usr/bin/which" .. cmd)
2. 编辑/usr/lib/lua/luci/model/diskman.lua， 在约19行下面增加：d.command.df = nixio.fs.access("/usr/bin/df") and "/usr/bin/df" or "/bin/df"

yunhaivip, 2024-10-16:
同样问题，arm平台，以上方法都没有用，原创大神不修复一下吗